tscircuit/pcb-viewer #36: Work on the buldge for the `trace_width` with large value

imrishabh18, 2024-08-15:
This is how it looks for the dramatic 20 width 🤔. Now I get what you mean't by that buldge

[image]

_Originally posted by @imrishabh18 in https://github.com/tscircuit/pcb-viewer/issues/35#issuecomment-2292214892_

seveibar, 2024-08-15:
hehe ok something was a tiny bit off, well we're a bit closer now at least.

imrishabh18, 2024-08-15:
Which line? I have three route endpoints and all are there

seveibar, 2024-08-16:
[image]

I think the first line is missing

imrishabh18, 2024-08-16:
The width is off, I think because the default value of the thickness of the route is 1 (when no width is mentioned)

seveibar, 2024-08-16:
The default has a couple places where it can come from- the Board (from <board /> should set the default in builder.

the shape isnt right though, it should narrow at the point

[image]

seveibar, 2024-08-16:
There should be a default defined in the builder but i think its supposed to be 0.05 (but we shouldn't hard code that anywhere except as the board/build context default)

seveibar, 2024-08-16:
[image]

Another example of how the widths should change

closed 2024-08-31 by imrishabh18