tscircuit/pcb-viewer #84: Silkscreen lines don't move when you use the manual edits prop

seveibar, 2024-12-07:
[image]

seveibar, 2024-12-07:
/bounty $20

ShiboSoftwareDev, 2024-12-09:
/attempt #84 i think i found a fix

buildingvibes, 2026-02-10:
/attempt #84

## Implementation Plan

The issue is that silkscreen lines (and other silkscreen elements) don't move when their parent component is moved via manual edit events.

### Root Cause
The `applyEditEvents` function from `@tscircuit/core` applies transformations to elements with matching `pcb_component_id`, but the `transformPCBElement` utility doesn't properly handle silkscreen line coordinates (x1, y1, x2, y2).

### Solution
I will create a wrapper function `applyEditEventsWithSilkscreenFix` that:
1. Calls the standard `applyEditEvents` function
2. Post-processes the result to apply the same translation to silkscreen elements
3. Handles all silkscreen element types:
   - pcb_silkscreen_line (x1, y1, x2, y2)
   - pcb_silkscreen_path (route array)
   - pcb_silkscreen_circle (center)
   - pcb_silkscreen_rect (center)
   - pcb_silkscreen_oval (center)
   - pcb_silkscreen_pill (center)
   - pcb_silkscreen_text (anchor_position)

### Files Modified
- `src/PCBViewer.tsx`: Update to use the new wrapper function
- `src/lib/apply-edit-events-with-silkscreen-fix.ts`: New utility function
- `src/examples/2025/silkscreen/silkscreen-line-with-component-move.fixture.tsx`: Test fixture

closed 2026-03-06 by Abse2001