tscircuit/pcb-viewer #92: Implement Rotations when moving components

seveibar, 2024-12-10:
After a component is selected, it should have a small circle in the right hand corner that can be used to rotate it. This will require emitting and applying a new type of edit event that rotates

ShiboSoftwareDev, 2024-12-11:
@seveibar how do you imagine the button to look like? you hold and spin it around the component to rotate it?

UTK-a-RSH, 2024-12-12:
Can I get assigned this issue please ?

Bhavyajain21, 2024-12-12:
/attempt #92

Bhavyajain21, 2024-12-12:
Can I be assigned? @seveibar

ShiboSoftwareDev, 2024-12-12:
@Bhavyajain21 @UTK-a-RSH we don't assign bounties except for time consuming ones

Bhavyajain21, 2024-12-13:
> [@Bhavyajain21](https://github.com/Bhavyajain21) [@UTK-a-RSH](https://github.com/UTK-a-RSH) we don't assign bounties except for time consuming ones

Do we have a discord channel for tscircuit?

ShiboSoftwareDev, 2024-12-13:
@Bhavyajain21 https://discord.gg/RvyP4V9Q

makaiachildress-web, 2026-02-17:
/attempt #92

closed 2026-02-17 by Abse2001