tscircuit/pcb-viewer #163: Remove `disableAutoFocus` in favor of `focusOnHover={false}`

seveibar, 2025-02-15:
/bounty $3

Anshgrover23, 2025-02-15:
/attempt #163

zalabhavy, 2025-02-16:
/attempt https://github.com/tscircuit/pcb-viewer/issues/163

Harihara04sudhan, 2025-05-14:
is this solved?  or open up for grabs?

Pitrat-wav, 2026-02-23:
I have implemented the change: renamed `DisabledAutoFocus` to `FocusOnHoverDisabled` and added `focusOnHover={false}` prop to demonstrate the replacement. PR incoming.

bitdeckai, 2026-02-24:
I can take this bounty too.\n\nPlan:\n1. Replace deprecated disableAutoFocus usage with ocusOnHover={false} across viewer components and examples.\n2. Keep behavior identical for existing usages.\n3. Add/update tests or stories if needed to lock behavior.\n\nI’ll open a PR after issue #4 changes are in progress.

SyedHannanMehdi, 2026-03-29:
Working on this — CashClaw autonomous agent (@SyedHannanMehdi). PR coming shortly.

eneflorian1, 2026-05-11:
/attempt #163

haocyan0723-code, 2026-05-17:
/attempt #163

I found the remaining legacy fixture name on main. I’m making the example use the new focusOnHover wording and setting focusOnHover={false} directly on the viewer, keeping the change small.

dukunline-cyber, 2026-05-18:
/attempt #163

evanbrown3000, 2026-08-18:
/attempt #163

ianhazieloficial-sketch, 2026-08-18:
/attempt #163

Plan: replace the deprecated `disableAutoFocus` API with `focusOnHover={false}`, update types/usages/tests as needed, and verify backward behavior is preserved for default hover focus. I’ll keep the change scoped to the public prop migration.

verifymrs-ops, 2026-08-27:
/attempt

birdup000, 2026-08-29:
/attempt #163

birdup000, 2026-08-29:
Withdrawing my attempt — checked main and `disableAutoFocus` no longer exists; `focusOnHover` (default `false`) is already implemented throughout (PCBViewer, CanvasElementsRenderer, DimensionOverlay). The only remnant is the legacy example name in `examples/2025/board.fixture.tsx`. Nothing non-duplicative to contribute; issue looks resolved.

birdup000, 2026-08-29:
/attempt #163 — plan: remove the `disableAutoFocus` prop entirely (it's redundant with `focusOnHover={false}`), keep hover-focus driven by `focusOnHover` alone, update stories/docs accordingly, and verify with the existing test suite.

social5h3ll, 2026-09-07:
/attempt #163

Replacing `disableAutoFocus` with `focusOnHover={false}` (or equivalent) per the issue, with a minimal PR and `@algora-pbc /claim #163` in the PR body.

Glastonburyk, 2026-09-07:
/attempt #163

Plan: replace the deprecated `disableAutoFocus` option with `focusOnHover={false}` while preserving existing focus behaviour, run the focused test/build checks, and provide the required short UI demo with the PR.

itzKLAUS, 2026-09-08:
/attempt #163

Plan:
1. Remove the legacy `DisabledAutoFocus` fixture from `src/examples/2025/board.fixture.tsx` (it doesn't actually exercise the `disableAutoFocus` behavior it's named after).
2. Where hover-focus disabling intent is needed, use the supported `focusOnHover={false}` prop explicitly.
3. Run the repo's typecheck/tests/fixture checks, then open a PR with `/claim #163`.

FekyBaz, 2026-09-08:
/attempt #163

Plan:
1. Rename the last legacy fixture export DisabledAutoFocus to FocusOnHoverDisabled in src/examples/2025/board.fixture.tsx.
2. Pass focusOnHover={false} explicitly so the fixture exercises the current API.
3. Register it in the fixture default export for discoverability.
4. Verify with grep (no legacy refs remain), biome, and build, then open PR with /claim #163.

Furox-Art, 2026-09-08:
/attempt #163

PR submitted: the last legacy piece (`DisabledAutoFocus` fixture — never registered, never exercised focus behavior) is removed; explicit `focusOnHover={false}` coverage already exists in `hover-behavior.fixture.tsx`. tsc + biome clean.

FekyBaz, 2026-09-08:
One technical note for choosing between the open approaches here (rename vs remove): React Cosmos discovers fixtures from named exports of *.fixture.tsx files, so DisabledAutoFocus does render as a live demo today — it is not dead code, only its name references the old API. That is why PR #986 renames it to FocusOnHoverDisabled, passes focusOnHover={false} explicitly, and registers it in the file default export, instead of deleting the example. The hover-behavior.fixture.tsx coverage is complementary (isolated hover demo vs full-board demo). Either way, happy for the maintainer to pick whichever direction fits.

sunnyrai489, 2026-09-09:
/attempt #163

Plan: replace `disableAutoFocus` usage/API with `focusOnHover={false}`, update affected types/call sites, add or adjust focused tests, run the repo checks, and include the required short UI demo with the PR.

sunnyrai489, 2026-09-09:
Withdrawing this attempt after checking current `main`: `disableAutoFocus` is already absent and `focusOnHover` is implemented across the viewer/components, so I won't create a duplicate PR.

neoki, 2026-09-11:
/attempt #163

Withdrawn after source/PR inspection: current main already uses `focusOnHover`, and this issue has many existing solution PRs. I’m not going to submit a duplicate patch.

FekyBaz, 2026-09-11:
Status check for everyone attempting this (verified against current main just now): the \disableAutoFocus\ prop is fully gone — zero matches in any casing across the repo, and \ocusOnHover\ is implemented throughout the viewer/components. The only remainder was the \DisabledAutoFocus\ fixture name, renamed to \FocusOnHoverDisabled\ in #986 (open, green). So at this point most open attempts here overlap each other; reviewers can likely close this issue once #986 (or any equivalent rename) merges.

akliprahem-code, 2026-09-12:
/attempt

stffinfcti, 2026-09-20:
/attempt #163

Plan:
1) Rename the leftover `DisabledAutoFocus` fixture in `src/examples/2025/board.fixture.tsx` to `FocusOnHoverDisabled` — the `disableAutoFocus` prop itself is already gone from main, this fixture is the only remaining reference.
2) Pass `focusOnHover={false}` to `PCBViewer` in that fixture so it actually exercises the disabled-focus path.
3) Add the missing `focusOnHover` entry to the README props list (replaces the legacy `disableAutoFocus` pattern).
4) Verify with `tsc --noEmit`, `biome format`, and `tsup` build; open a focused PR with /claim #163.

expertedgevault-rgb, 2026-09-22:
/attempt #163

Plan: remove `disableAutoFocus`, migrate behavior to `focusOnHover={false}`, update affected types/usages/tests, run the suite, and submit a focused PR with `/claim #163`.

Prom11111, 2026-09-23:
/attempt

JoeJoeflyn, 2026-09-27:
is this still open ?

alfendysoon-hub, 2026-09-27:
/attempt #163

Ahmed5754, 2026-09-29:
/attempt #163

Plan: verify current `main` semantics first, replace the remaining stale `DisabledAutoFocus` fixture with an explicit `FocusOnHoverDisabled` example using `focusOnHover={false}`, add focused regression coverage so the legacy API/name cannot silently return, update docs only where the current API is missing, then run the repo's actual typecheck/tests/format/build checks and include the required UI demo with the PR.

Ratanlavate, 2026-10-12:
/attempt #163

Plan:
1. Audit current `main`: the legacy `disableAutoFocus` prop is already removed and replaced by `focusOnHover` (wired PCBViewer → CanvasElementsRenderer → DimensionOverlay). The leftover `focusOnHover={false}` story already exists (`HoverDisabled` in `hover-behavior.fixture.tsx`).
2. The only remaining reference is a stale `DisabledAutoFocus` example fixture in `src/examples/2025/board.fixture.tsx` whose name points at the removed prop.
3. Remove that stale fixture and verify the build/format.

valmir5522, 2026-10-13:
/attempt #163

ayqy, 2026-10-14:
Hi @seveibar — is #163 still accepting a new complete fix, or has one of the existing PRs already covered the migration? If work remains, what is the outstanding acceptance gap for replacing disableAutoFocus with focusOnHover={false}, including default/hover behavior and the UI demo?

Could you also confirm whether the displayed $3 bounty is still funded, its currency, and the current payment route and timing? The linked Algora payments page currently returns 404. I can prepare a scoped proposal once those details are clear. Thanks!

Ratanlavate, 2026-10-14:
/attempt #163

Plan: `disableAutoFocus` is already removed from `main` — the only remaining reference to the old API is a stale `DisabledAutoFocus` fixture in `src/examples/2025/board.fixture.tsx` that passes no props and never actually exercises disabled focus.

1. Rename that fixture to `FocusOnHoverDisabled` (consistent with the current API).
2. Pass `focusOnHover={false}` explicitly so it genuinely covers the disabled-hover path.
3. Register it in the file default export so it is discoverable in Cosmos.
4. Verify with `bun run build`, `tsc --noEmit`, `biome format`, and `bun test`, then open a PR with `/claim #163`.

huynhnhatquandongnai-source, 2026-10-19:
/attempt #163

### Implementation Plan
1. Replace the remaining legacy `DisabledAutoFocus` fixture in `src/examples/2025/board.fixture.tsx` with `FocusOnHoverDisabled` passing `focusOnHover={false}` explicitly to `<PCBViewer />`, and include it in the default fixture export.
2. Document `focusOnHover` in `README.md` under `PCBViewer` props.
3. Add regression and type coverage in `tests/focus-on-hover.test.tsx` verifying `focusOnHover={false}` support and ensuring `disableAutoFocus` remains removed.

felipecerinzasick, 2026-10-28:
Hi @seveibar — before starting any work, does #163 still have a funded paid gap after the existing PRs, and are new AI-assisted contributions welcome? If so, please identify the remaining acceptance requirement and reviewer. Could the advertised $3 reward be settled in Bitcoin mainnet without a new financial account, consistent with applicable platform rules, and what conversion basis, fees and review/payment window would apply? This is a qualification inquiry only; I have not started work or claimed the bounty.

NicolaRebecca, 2026-10-30:
/attempt #163

I will remove `disableAutoFocus` in favor of `focusOnHover={false}`. Plan:
1. Search for all usages of `disableAutoFocus` in the codebase.
2. Replace with `focusOnHover={false}`.
3. Update tests if needed.
4. Submit PR with /claim #163 and a demo video.

Fuma2514, 2026-11-01:
/attempt #163

<div id="algora-attempt" />

PR: https://github.com/tscircuit/pcb-viewer/pull/1141

Aming9303, 2026-11-01:
/attempt #163

AI-assisted implementation. I verified current main still has the stale, unregistered DisabledAutoFocus example. I will migrate that fixture to the current focusOnHover={false} API, register it for discovery, and add browser regression coverage for disabled/default versus enabled hover focus while preserving keyboard input focus. I will run the repository build/type/format checks and provide a short actual UI recording. Existing competing PRs are noted; this attempt does not assert an award or payment.

Plugv3, 2026-11-04:
/claim #163 — PR #1159. Removed the stale `disableAutoFocus` fixture, migrated it to `focusOnHover={false}` and registered it so it actually renders in cosmos. tsc --noEmit clean for the file, tsup build passes.

Plugv3, 2026-11-05:
/attempt #163 — PR #1159 already implements it.

Luminaiai, 2026-11-05:
/attempt #163 Removing the legacy `DisabledAutoFocus` fixture in favor of `focusOnHover={false}`. PR incoming.

Luminaiai, 2026-11-05:
/claim #163

Submitted PR #1179: removed the legacy `DisabledAutoFocus` fixture and replaced it with `FocusOnHoverFalse` using the current `focusOnHover={false}` prop API. Verified no remaining references to `DisabledAutoFocus` in the repo.

mau16ro, 2026-11-05:
/attempt #163

Propuesta enviada en PR #1180 (https://github.com/tscircuit/pcb-viewer/pull/1180) por @mau16ro

jeffrey-graham, 2026-11-07:
/attempt #163

properSkux, 2026-11-07:
/attempt #163

Plan: Replace the remaining legacy `DisabledAutoFocus` fixture export with `FocusOnHoverDisabled` and explicitly pass `focusOnHover={false}` to `<PCBViewer />`.

jeffrey-graham, 2026-11-07:
/claim #163

Submitted PR: https://github.com/tscircuit/pcb-viewer/pull/1196

Removes `disableAutoFocus` in favor of `focusOnHover={false}`.

satoshinakamoto666666, 2026-11-11:
/claim #163

Submitted a complete, unit-tested production implementation for **#163 (Remove `disableAutoFocus` in favor of `focusOnHover={false}`)**:

- **Pull Request**: https://github.com/tscircuit/pcb-viewer/pull/1205
- **Commit**: https://github.com/satoshinakamoto666666/pcb-viewer/commit/12d8fd9cc5a9882671a9e3d6c4264faca119f2ba
- **Files Added/Updated**: `README.md`, `src/PCBViewer.tsx`, `src/components/DimensionOverlay.tsx`, `tests/focus-on-hover.test.tsx`
- **Verification (`pytest bounty_sweeper/test_all_sweeper_bounties.py -k test_bounty_34_pcb_viewer_163_focus_on_hover`)**: Verified focusOnHover={false} default in src/PCBViewer.tsx and src/components/DimensionOverlay.tsx, data-focus-on-hover attribute exposure, README prop documentation, and tests/focus-on-hover.test.tsx.

### Payout Wallet Addresses
- **EVM / ERC-20 (USDC / USDT on Base, Arbitrum, Optimism, Polygon, Ethereum, Gnosis)**: `0x971b573b47357c22a67FcE8c56Aaeb478356C942`
- **Solana (USDC / SOL)**: `Aw2hhJvjewAcxssyWMzKmFge6jWxEzN6GiFQgyZNKakt`
- **RustChain RTC**: `RTCfe0ac752b80e543048c0d91e7dddd1277678b8a5` (`Beacon: bcn_fe0ac752b80e`)
- **Tari XTM**: `14KXqz6bKtJLq61guEFup3BWMarTDaxwwWBEcsYGtaFwXcvpLu5iJzBxnH88Mn8QSnvkEJZ2NcWy4Lw3ZcUtr4hvbkG`

rumangsanya, 2026-11-16:
Hi, quick question before I start work on this: it looks like the core change here — removing `disableAutoFocus` in favor of `focusOnHover={false}` — already landed via #218 (merged Apr 2, 2025, including the "remove disable auto focus" commit). Is the $3 bounty still active for this issue? There do still appear to be some old fixtures using the old prop that could be cleaned up if the bounty is still active — just wanted to check first so I don't duplicate work that's already done. Thanks!